sparkfun/SparkFun_BME280_Arduino_Library #1: Reference pressure can't be set

blenheim, 2015-11-16:
Dear Sir, my name is Emanuele and I'am an amateur softwarist. I was trying your library and I found many interesting features including the possibility to use oversampling.
The only problem I've found is that reference pressure cannot be set for proper altitude calculations. I've seen in the code, under Pressure Section that the reference pressure is set at 101325Pa; this value should be changed accordingly to the real pressure at sea level in order to calculate the right altitude.
Hope this can be useful.
Best regards,
Emanuele

nalanbar, 2017-03-18:
This was very helpful. It would be nice if the library could be updated to allow us to set a reference pressure more easily than modifying SparkFunBME280.cpp. 

That said, if anyone else needs to solve this themselves, the line you need to update is 190. I've been getting my local airport's METAR, converting the InMG to Pa, and then updating it in SparkFunBME280.cpp. It's not a great solution, but it at least get's my altitude above sea level. 😄

nseidle, 2018-03-23:
This is an excellent idea. I've exposed a function called setReferencePressure() in the latest version. Please have a test when you get the chance. I will try to include an example as well.

Lines changed: [function](https://github.com/sparkfun/SparkFun_BME280_Arduino_Library/blob/master/src/SparkFunBME280.cpp#L306), [calculation](https://github.com/sparkfun/SparkFun_BME280_Arduino_Library/blob/master/src/SparkFunBME280.cpp#L323).

closed 2018-03-23 by nseidle