sparkfun/SparkFun_BME280_Arduino_Library #4: Calibration's problem

QwiateQ, 2016-01-23:
Hello,

I found an problem reading data from BME280. Where should I look and search, according to an fact that on some other library's data are readable on Arduino monitor.
I2C_ReadAllData.ino respond as below:

Program Started
Starting BME280... result of .begin(): 0x1
Displaying ID, reset and ctrl regs
ID(0xD0): 0x1
Reset register(0xE0): 0x1
ctrl_meas(0xF4): 0x1
ctrl_hum(0xF2): 0x1

Displaying all regs
0x80:80 81 82 83 84 85 86 87 88 89 8A 8B 8C 8D 8E 8F 
0x90:90 91 92 93 94 95 96 97 98 99 9A 9B 9C 9D 9E 9F 
0xA0:A0 A1 A2 A3 A4 A5 A6 A7 A8 A9 AA AB AC AD AE AF 
0xB0:B0 B1 B2 B3 B4 B5 B6 B7 B8 B9 BA BB BC BD BE BF 
0xC0:C0 C1 C2 C3 C4 C5 C6 C7 C8 C9 CA CB CC CD CE CF 
0xD0:D0 D1 D2 D3 D4 D5 D6 D7 D8 D9 DA DB DC DD DE DF 
0xE0:E0 E1 E2 E3 E4 E5 E6 E7 E8 E9 EA EB EC ED EE EF 
0xF0:F0 F1 F2 F3 F4 F5 F6 F7 F8 F9 FA FB FC FD FE FF 

Displaying concatenated calibration words
dig_T1, uint16: 257
dig_T2, int16: 257
dig_T3, int16: 257
dig_P1, uint16: 257
dig_P2, int16: 257
dig_P3, int16: 257
dig_P4, int16: 257
dig_P5, int16: 257
dig_P6, int16: 257
dig_P7, int16: 257
dig_P8, int16: 257
dig_P9, int16: 257
dig_H1, uint8: 1
dig_H2, int16: 257
dig_H3, uint8: 1
dig_H4, int16: 17
dig_H5, int16: 16
dig_H6, uint8: 1

Temperature: 0.00 degrees C
Temperature: 32.00 degrees F
Pressure: 21742956.00 Pa
Altitude: -81480.92m
Altitude: -267325.84ft
%RH: 0.00 %

Temperature: 0.00 degrees C
Temperature: 32.00 degrees F
Pressure: 21742956.00 Pa
Altitude: -81480.92m
Altitude: -267325.84ft
%RH: 0.00 %

I'll be glad for some help.
Best regards,
QwiateQ

marshalltaylorSFE, 2016-01-25:
This shows that communication never occurred with the BME280 from the start.  The ID should be something like 0x62, not 1.  I would check the connection lines.  The sketch defaults to I2C, address 0x77, on uno pins A4 (SDA) and A5(SCL).  Make sure those connections are made, plus connection of ground and 3.3v to VCC.

What micro controller are you using?

nseidle, 2018-03-23:
Looks like you got if fixed? Closing out issue. Please reopen if you continue to have problems! We're here to help.

closed 2018-03-23 by nseidle