sparkfun/SparkFun_BME280_Arduino_Library #6: High current consumption

PaulRB, 2016-03-18:
Reading the temp/pressure/humidity using this library seems to leave the bme280 in a state where it consumes over 400uA, and stays like that indefinitely. It would be better if the readTempC(), readFloatHumidity() and readFloatPressure() functions put the chip into sleep or standby mode after the reading has been taken. These modes use < 1uA. See page 7 of the data sheet: https://www.adafruit.com/datasheets/BST-BME280_DS001-10.pdf

Workaround: user can call reset() after taking readings. This means that begin() must be called before readings are taken, e.g.:
`  combinedSensor.begin();
  temperatureNow = combinedSensor.readTempC();
  double humidityNow = combinedSensor.readFloatHumidity();
  double pressureNow = combinedSensor.readFloatPressure() / 100;
  combinedSensor.reset();`
This reduces current consumption to around 8uA vs. 400uA after readings are taken, according to my multimeter.

PaulRB, 2016-03-18:
Just realised there is a pull request to add a forceReadings() function which sounds like it may correct this issue.

JamesWebb73, 2016-03-19:
I suspect most users (well me anyway) will want to use the _continuous_ mode to reduce measurement noise, and not worry about current.

PaulRB, 2016-03-20:
Reading a few more pages from the data sheet, the different use scenarios and recommended modes are discussed. My uses are weather & humidity monitoring. I assume yours would be movement detection & gaming. Clearly Bosch recognised these differences. It would be nice if the library was enhanced to allow their selection.

ripper121, 2016-09-19:
Did you get it working with lower current?
I like to build a environment sensor and need really low current consumption for this.
Maybe you can help me to set the BME280 correct.

nseidle, 2018-03-24:
We've added setMode(MODE_FORCED) as well as a low power example to the library v2.0. I'm not entirely confident in the mode setting description in the BME280 datasheet. We believe the "Example 6 Low Power" works but it's a bit disconcerting that the measurement bit (status bit 3) doesn't go to zero as soon as mode is set to forced (0b01). Please have a look and let us know if anything seems out of place.

closed 2018-03-24 by nseidle