sparkfun/SparkFun_BME280_Arduino_Library #19: BME280 GYBMEP incorrect sensor readings

mailgpa, 2018-03-09:
Hi,

I am trying to read sensor data from BME280 GYBMEP connected to ESP8266 (MOD-WIFI-ESP8266-DEV by Olimex) via I2C. There is also APDS-9960 light sensor connected in parallel.

Readings from BME280 make no sense at all (see below). Is it some kind of calibration issue or a faulty sensor?

```
SDK:2.2.1(cfd48f3)/Core:2.4.1/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1)
APDS-9960 initialization complete
Light sensor is now running
Starting BME280... result of .begin(): 0xD0
Displaying ID, reset and ctrl regs
ID(0xD0): 0xD0
Reset register(0xE0): 0xD0
ctrl_meas(0xF4): 0xD0
ctrl_hum(0xF2): 0xD0

Displaying all regs
0x80:08 08 08 08 08 08 08 08 08 08 08 08 08 08 08 08 
0x90:09 09 09 09 09 09 09 09 09 09 09 09 09 09 09 09 
0xA0:0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 0A 
0xB0:0B 0B 0B 0B 0B 0B 0B 0B 0B 0B 0B 0B 0B 0B 0B 0B 
0xC0:0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 0C 
0xD0:0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 0D 
0xE0:0E 0E 0E 0E 0E 0E 0E 0E 0E 0E 0E 0E 0E 0E 0E 0E 
0xF0:0F 0F 0F 0F 0F 0F 0F 0F 0F 0F 0F 0F 0F 0F 0F 0F 

Displaying concatenated calibration words
dig_T1, uint16: 43728
dig_T2, int16: -21808
dig_T3, int16: -21808
dig_P1, uint16: 43728
dig_P2, int16: -21808
dig_P3, int16: -21808
dig_P4, int16: -21808
dig_P5, int16: -21808
dig_P6, int16: -21808
dig_P7, int16: -21808
dig_P8, int16: -21808
dig_P9, int16: -21808
dig_H1, uint8: 208
dig_H2, int16: -21808
dig_H3, uint8: 208
dig_H4, int16: -21248
dig_H5, int16: -21075
dig_H6, uint8: 208

Init completed!
Tick...
Light: Ambient: 0 R: 0 G: 0 B: 0
Temperature: 50.30 degrees C
Pressure: 5522.10 Pa
Altitude: 19489.89m
%RH: 0.00 %

Tick...
Light: Ambient: 166 R: 55 G: 63 B: 64
Temperature: 50.30 degrees C
Pressure: 5522.10 Pa
Altitude: 19489.89m
%RH: 0.00 %

Tick...
Light: Ambient: 166 R: 55 G: 63 B: 64
Temperature: 50.30 degrees C
Pressure: 5522.10 Pa
Altitude: 19489.89m
%RH: 0.00 %

Tick...
Light: Ambient: 166 R: 55 G: 63 B: 64
Temperature: 50.30 degrees C
Pressure: 5522.10 Pa
Altitude: 19489.89m
%RH: 0.00 %

Tick...
Light: Ambient: 165 R: 55 G: 63 B: 64
Temperature: 50.30 degrees C
Pressure: 5522.10 Pa
Altitude: 19489.89m
%RH: 0.00 %

Tick...
Light: Ambient: 165 R: 54 G: 63 B: 64
Temperature: 50.30 degrees C
Pressure: 5522.10 Pa
Altitude: 19489.89m
%RH: 0.00 %

Tick...
Light: Ambient: 165 R: 54 G: 63 B: 64
Temperature: 50.30 degrees C
Pressure: 5522.10 Pa
Altitude: 19489.89m
%RH: 0.00 %

Tick...
Light: Ambient: 165 R: 54 G: 63 B: 64
Temperature: 50.30 degrees C
Pressure: 5522.10 Pa
Altitude: 19489.89m
%RH: 0.00 %

Tick...
Light: Ambient: 165 R: 55 G: 63 B: 64
Temperature: 50.30 degrees C
Pressure: 5522.10 Pa
Altitude: 19489.89m
%RH: 0.00 %

Tick...
Light: Ambient: 165 R: 55 G: 63 B: 64
Temperature: 50.30 degrees C
Pressure: 5522.10 Pa
Altitude: 19489.89m
%RH: 0.00 %
```

mailgpa, 2018-03-09:
Ok, I've found an issue. It's different I2C address in my case - 0x76

closed 2018-03-09 by mailgpa

nseidle, 2018-03-09:
Thanks for coming back to let us know!