sparkfun/SparkFun_BME280_Arduino_Library #21: sensor on i2c with 0x76 address doesn't work since update

LukaGitH, 2018-03-24:
It stopped working with 2.0, works fine in 1.2

nseidle, 2018-03-25:
Hi Luka - *Thank you* for checking my work! Please have a look at the latest commit. I've updated example 2 (I was trying to do too much in one example). I've tested on both RedBoard and Teensy and should be good to go. 

I've also added an advanced example 9 that uses software I2C. It's a bit of a hack job, and requires the user to edit the SparkFunBME280.h file but I'd welcome your feedback.

Additionally, I've added a basic SPI example.

Once you give me the green light I'll push a release so the library manager will increment.

LukaGitH, 2018-03-25:
The problem is still here, sensor witn 0x76 can't be read. Even the old I2C_and_SPI_Multisensor which for mySensorB changed to:

mySensorB.settings.commInterface = I2C_MODE;
mySensorB.settings.I2CAddress = 0x76;

Is not working, while in 1.2 version it works fine. Therefor I think the problem is library still

Program Started
Starting BME280s... result of .begin():
Sensor A: 0x60
Sensor B: 0x58
Temperature: 21.91, 0.00 degrees C
Temperature: 71.44, 32.00 degrees F
Pressure: 95971.60, 0.00 Pa
Altitude: 470.81, 45846.20m
Altitude: 1544.65, 150414.04ft

nseidle, 2018-03-25:
What platform are you compiling on? What platform are you compiling for?

LukaGitH, 2018-03-25:
Arduino, pro mini at this point

nseidle, 2018-03-25:
It works fine for me. Pro Mini, 3.3V @ 8MHz. I am using the BME280 on our [Combo board](https://www.sparkfun.com/products/14348).

[image]

What board are you using? If you're using our [BME280 breakout](https://www.sparkfun.com/products/13676) did you close the jumper on the back of the board?

nseidle, 2018-03-26:
Neat! Ok, have you pulled the SDO pin to ground, before powering the device, to set the address to 0x76?

LukaGitH, 2018-03-26:
Yes, SDO is always ground on one, while Vcc on other. Like I said, hardware didn't change, only software did.

nseidle, 2018-03-27:
Hmm. [This delay](https://github.com/sparkfun/SparkFun_BME280_Arduino_Library/blob/master/src/SparkFunBME280.cpp#L64) was delay(10) in the original library. Please try changing that from 2 to 10 and see if it fixes your problem?

LukaGitH, 2018-03-27:
I think I have worked it out. It is not delay, since there was no delay before. Delay can stay at 2ms.
What the problem is, is this: 
```
BME can be 0x76 or 0x77
BMP can be only 0x77
```

BME has ID 0x60, BMP has ID 0x58 (mass production, samples have different ID)

in CPP, you check for 
`if(chipID != 0x60) return(chipID); //Failed!`
Which fails in case of BMP, so to fix this, this needs to be changed to
```
if(chipID != 0x60) // This is not BME
if(chipID != 0x58) // This is not BMP
return(chipID); //Failed!
```
So does 
`if(begin() == 0x60) return(true); //Begin normal init with these settings. Should return chip ID of 0x60
	return(false);`

to
if(begin() == 0x58) return(true); //Begin normal init with these settings. Should return chip ID of 0x58
if(begin() == 0x60) return(true); //Begin normal init with these settings. Should return chip ID of 0x60
	`return(false);`

LukaGitH, 2018-03-28:
No, my BMP are mass production, 0x58
From datasheet, BME doesn't samples. 

Sure! It's my first time, hope I get it right

nseidle, 2018-03-31:
Ah, now I understand. This library is written for the BME280, you're using the BMP280. If you want to submit a PR to allow the library to work with both BME and BMP I'm happy to have a look. I'm closing this because this is not really an issue with the library.

closed 2018-03-31 by nseidle

sonybansi, 2020-04-21:
HW-611difficult to figure out which sense is this.
I think this is a fake BMP280.
What do you think 🤔
Can this read humidity?

AndyEngland521, 2020-04-21:
@sonybansi Can you clarify what sensor you think is fake? Also, this library is for the BME280 instead of the BMP...

sonybansi, 2020-04-21:
@AndyEngland521 
> 

> HW-611difficult to figure out which sense is this.
> I think this is a fake BMP280.
> What do you think 🤔
> Can this read humidity?
> 
> [image]

AndyEngland521, 2020-04-21:
Oh, the picture wasn't showing up for me for some reason. I am not sure which sensor you have, read the chipid as described above to determind which sensor you have.