sparkfun/SparkFun_BME280_Arduino_Library #27: Initial reading inaccurate when using MODE_SLEEP

olicooper, 2018-09-19:
I `Serial.print()` the following to the console. Notice the intial `h_pc`, `p_hpa` and `alt_m` readings are off and how it takes 16ms to read...

```
Measure time 16ms => {"h_pc":51.34,"p_hpa":95007.30,"alt_m":558.15,"temp_c":21.32}
Measure time 8ms => {"h_pc":58.79,"p_hpa":98630.13,"alt_m":234.53,"temp_c":21.30}
Measure time 8ms => {"h_pc":51.03,"p_hpa":98629.98,"alt_m":234.55,"temp_c":21.26}
Measure time 8ms => {"h_pc":56.95,"p_hpa":98627.13,"alt_m":234.80,"temp_c":21.25}
Measure time 8ms => {"h_pc":53.36,"p_hpa":98624.58,"alt_m":235.02,"temp_c":21.25}
```

If it helps I am using **PlatformIO** with the latest ESP32 Arduino (1.3.0) on the **Lolin Wemos D1 mini Pro**

My `main.cpp` looks as follows:
```
#include <Arduino.h>
#include <Wire.h>
#include "SparkFunBME280.h"

BME280 bme; // I2C

void printValues();

void setup() {
    Serial.begin(9600);
    Serial.println(F("BME280 test"));

    Wire.begin(SDA,SCL);
    Wire.setClock(400000);
    bme.setI2CAddress(0x76);

    if (bme.beginI2C() == false) {
        Serial.println(F("No sensor response. Please check wiring."));
        ESP.deepSleep(0);
    }
    bme.setFilter(2); //0 to 4 is valid. Filter coefficient. See 3.4.4
    bme.setStandbyTime(5); //0 to 7 valid. Time between readings. See table 27.
    bme.setTempOverSample(1); //0 to 16 are valid. 0 disables temp sensing. See table 24.
    bme.setPressureOverSample(1); //0 to 16 are valid. 0 disables pressure sensing. See table 23.
    bme.setHumidityOverSample(1); //0 to 16 are valid. 0 disables humidity sensing. See table 19.
    bme.setMode(MODE_SLEEP); //MODE_SLEEP, MODE_FORCED, MODE_NORMAL is valid. See 3.3
}

void loop() {
    bme.setMode(MODE_FORCED); //Wake up sensor and take reading

    long startTime = millis();
    while(bme.isMeasuring() == false); //Wait for sensor to start measurment
    while(bme.isMeasuring() == true); //Hang out while sensor completes the reading    
    long endTime = millis();
    // bme.setMode(MODE_SLEEP); //Needed?

    //Sensor is now back asleep but we get get the data
    Serial.print(F("Measure time "));
    Serial.print(endTime - startTime);
    Serial.print("ms => ");

    printValues();
    delay(30e3); // 30 second sleep (30,000 ms)
}

void printValues() {
    Serial.print("{\"h_pc\":"); Serial.print(bme.readFloatHumidity(), 2);
    Serial.print(",");

    Serial.print("\"p_hpa\":"); Serial.print(bme.readFloatPressure(), 2);
    Serial.print(",");

    Serial.print("\"alt_m\":"); Serial.print(bme.readFloatAltitudeMeters(), 2);
    Serial.print(",");

    Serial.print("\"temp_c\":"); Serial.print(bme.readTempC(), 2);
    Serial.print("}\n");
}
```

danlewis, 2019-02-18:
I've been using forced mode and have also noticed the first reading can be far off for all values

pbolduc, 2019-11-05:
Initial readings for the humidity and pressure can be off based on your sample code because they depend on the temperature reading.  There is a private variable [t_fine](https://github.com/sparkfun/SparkFun_BME280_Arduino_Library/blob/7a7fb985268b7c8ca7678b1658b33d5ce541b980/src/SparkFunBME280.h#L170) that both [humidity](https://github.com/sparkfun/SparkFun_BME280_Arduino_Library/blob/7a7fb985268b7c8ca7678b1658b33d5ce541b980/src/SparkFunBME280.cpp#L457) and [pressure](https://github.com/sparkfun/SparkFun_BME280_Arduino_Library/blob/7a7fb985268b7c8ca7678b1658b33d5ce541b980/src/SparkFunBME280.cpp#L391) reference.  It is only initialized in when reading the [temperature](https://github.com/sparkfun/SparkFun_BME280_Arduino_Library/blob/7a7fb985268b7c8ca7678b1658b33d5ce541b980/src/SparkFunBME280.cpp#L490).   If you want more accurate humidity and pressure readings, I would advise reading the temperature first.

santaimpersonator, 2019-12-27:
@pbolduc is correct. The humidity and pressure sensor values are temperature compensated. The `t_fine` value would need to be updated by [`.readTempC()`](https://github.com/sparkfun/SparkFun_BME280_Arduino_Library/blob/master/src/SparkFunBME280.cpp#L461) function first; otherwise you will be using an "older value", which would affect your readings.

closed 2019-12-27 by santaimpersonator