sparkfun/SparkFun_BME280_Arduino_Library #30: Incorrect Coefficient in Altitude Calculation

seekerdave, 2018-11-17:
I believe there is an error in the altitude calculation formula (SparkFunBME280.cpp, line #421).  The coefficient value of **45846.2** does not match international (ISA) or US standard altitude models.

To match accepted standards this should have a value of **44330.77** (44330.76923 to be exact).  

I know it seems a small issue, but since this is a multiplier, it results in values that are 3.4% too high.  Here are some references supporting this coefficient:
 
https://www.iso.org/standard/7472.html
ISO Standard Atmosphere.  The primary source document for the international standard.  (Unfortunately there is a charge to download this document.)

https://www.fai.org/sites/default/files/documents/sc4_vol_edic_16_rev_1.pdf
FAI World Aeromodeling document.  A more accessable document describing altimeter requirements for model aircraft competition. (See page 10)

https://en.wikipedia.org/wiki/International_Standard_Atmosphere
Wikipedia info on altitude standards with more references at the end of the document.

https://github.com/adafruit/Adafruit_BME280_Library
The Adafruit BME280 library uses a value of 44330.0 (line 591).

Thank you for your efforts!

nseidle, 2018-12-10:
Thank you for reporting and the source links! Now fixed in v2.0.4.

closed 2018-12-10 by nseidle

ClaytonSmith, 2019-01-04:
What is the name of that coefficient?