sparkfun/SparkFun_BME280_Arduino_Library #31: Connecting to A4 & A5 doesn't work.

danlewis, 2019-02-18:
Worth noting that examples, tutorials and documentation made note to connect SDA and SCL to A4 and A5, however this did not work on my Arduino Uno or MKR Wifi 1010 - I had to connect them to the boards corresponding SDA and SCL inputs.

This initially caused me a little confusion and might be worth noting in examples.

nseidle, 2019-02-18:
Sorry, but I have to disagree with you. The pin labeled SDA on an Uno is electrically connected to pin labeled A4. Likewise with SCL(A5). There's no magic. They're the *same* thing. Just to check my sanity this morning:

[image]

Works like a charm.

I can't speak to the MKR Wifi board. Perhaps there's something else going on?

You make the point tho - the Uno R3 has been around for ages so we can stop telling people to use the A4/A5 pins and just use the SDA/SCL pins.

danlewis, 2019-02-19:
For the MKR Wifi 1010 it definitely didn't work on A4/A5 which was the main board I was using for it. Unfortunately my BME280 sensor died last week after only a couple of weeks of use so I can't retest it on my Arduino Uno Rev3 so thanks for double checking.

Cheers

nseidle, 2019-02-19:
Ah - indeed. I'll start pushing our docs to read SDA/SCL.

closed 2019-02-19 by nseidle