sparkfun/SparkFun_BME280_Arduino_Library #32: Example9_SoftwareI2C/Example9_SoftwareI2C.ino:38: undefined reference to `BME280::beginI2C(SoftwareWire&)'

supersjimmie, 2019-03-17:
I took the exact Example9_SoftwareI2C.ino sample without any modification, compile fails with:
```
C:\Users\me\AppData\Local\Temp\cc9OHCNm.ltrans0.ltrans.o: In function `setup':

C:\Users\me\Documents\arduino-1.8.8\libraries\SparkFun_BME280_Arduino_Library-master\examples\Example9_SoftwareI2C/Example9_SoftwareI2C.ino:38: undefined reference to `BME280::beginI2C(SoftwareWire&)'

collect2.exe: error: ld returned 1 exit status```

supersjimmie, 2019-03-17:
Did I need to change these in the .h file?
```
#include <SoftwareWire.h> //SoftwareWire by Testato. Installed from library manager.
#define I2C_MODE 2

AndyEngland521, 2019-11-18:
@supersjimmie You do, check out the comment in the beginning of the example, it states that you'll need to edit the .h file for the example to work.

closed 2019-11-18 by AndyEngland521