sparkfun/SparkFun_BME280_Arduino_Library #33: Support Custom Wire Ports

oclyke, 2019-08-09:
According to [this SparkFun forums post](https://forum.sparkfun.com/viewtopic.php?f=165&t=50601&p=206724#p206724) the library does not support additional wire ports like Wire1, Wire2 etc... 

This is a useful feature for many arduino-compatible boards with addtl. capabilities.

AndyEngland521, 2019-11-18:
Are you sure? As far as I can tell we're using TwoWire and a custom Wire Port in the beginI2C() function which is what we do everywhere else. The confusion probably stems from the fact that this device is both I2C and SPI and contains begin(), beginSPI(), and beginI2C() functions. Overloading the first begin function to also take on a wire port/chip select pin wasn't done at the outset of the creation of this library, but it looks like the option is there.

closed 2019-11-18 by AndyEngland521

oclyke, 2019-11-18:
Usage of ```beginI2C()``` is illustrated in ```Example9_SoftwareI2C```