sparkfun/SparkFun_BME280_Arduino_Library #38: Error compiling for board SparkFun SAMD51 Thing Plus

kisom, 2019-10-28:
I wanted to hook up my lovely SparkFun BME280 / CCS811 Qwiic board to my SAMD51 Thing Plus (for a solar-powered LoRaWAN environmental sensor :D), but ran into some issues compiling a sketch with this library.

I pared it down to this sketch, which I believe is the bare minimum needed to get the BME280 working:

```c++
#include <SparkFunBME280.h>

BME280  bme280;

void setup() {
  Serial.begin(9600);
  while (!Serial) ;
  bme280.settings.commInterface = I2C_MODE;
  bme280.settings.I2CAddress = 0x77;
  bme280.settings.runMode = 3; //  3, Normal mode
  bme280.settings.tStandby = 0; //  0, 0.5ms
  bme280.settings.filter = 0; //  0, filter off

  //tempOverSample can be:
  //  0, skipped
  //  1 through 5, oversampling *1, *2, *4, *8, *16 respectively
  bme280.settings.tempOverSample = 1;

  //pressOverSample can be:
  //  0, skipped
  //  1 through 5, oversampling *1, *2, *4, *8, *16 respectively
  bme280.settings.pressOverSample = 1;

  //humidOverSample can be:
  //  0, skipped
  //  1 through 5, oversampling *1, *2, *4, *8, *16 respectively
  bme280.settings.humidOverSample = 1;

  if (bme280.begin() != 0x60) {
    Serial.println("BME280 failed");
    while (1);
  }

  Serial.println("OK");
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

But it doesn't compile:

```
Arduino: 1.8.9 (Linux), TD: 1.47, Board: "SparkFun SAMD51 Thing Plus"

Build options changed, rebuilding all
/home/kyle/Arduino/libraries/SparkFun_BME280/src/SparkFunBME280.cpp: In member function 'uint8_t BME280::begin()':
/home/kyle/Arduino/libraries/SparkFun_BME280/src/SparkFunBME280.cpp:98:23: error: 'SPI_CLOCK_DIV32' was not declared in this scope
   SPI.setClockDivider(SPI_CLOCK_DIV32);
                       ^
exit status 1
Error compiling for board SparkFun SAMD51 Thing Plus.
```

I've verified that this works with the RedBoard Turbo.

My investigation shows that the SparkFun SAMD core only defines the clock dividers for 48MHz CPUs; I'll try to follow on with a PR for that in that repo. In the meantime, I've added the following after the 48MHz definitions (in ~/.arduino15/packages/SparkFun/hardware/samd/1.7.0/libraries/SPI):

```
#if F_CPU == 120000000
  #define SPI_CLOCK_DIV2   15
  #define SPI_CLOCK_DIV4   30
  #define SPI_CLOCK_DIV8   60
  #define SPI_CLOCK_DIV16  120
  #define SPI_CLOCK_DIV32  240
  #define SPI_CLOCK_DIV64  480
  #define SPI_CLOCK_DIV128 960
#endif
```

where each value is (120 MHz / 16 MHz * divider), as with the block above it.

Cheers!

kisom, 2019-10-28:
I forgot to mention; adding this fix makes the sketch compile. Great success!

AndyEngland521, 2019-10-30:
fixed in SAMD v1.7.1 thanks to your PR, closing this issue :)

closed 2019-10-30 by AndyEngland521