sparkfun/SparkFun_BME280_Arduino_Library #39: reading pressure or humidity without temperature can result incorrect results

pbolduc, 2019-11-05:
The readings for the humidity and pressure can be off because they depend on the temperature reading.  There is a instance variable [t_fine](https://github.com/sparkfun/SparkFun_BME280_Arduino_Library/blob/7a7fb985268b7c8ca7678b1658b33d5ce541b980/src/SparkFunBME280.h#L170) that both [humidity](https://github.com/sparkfun/SparkFun_BME280_Arduino_Library/blob/7a7fb985268b7c8ca7678b1658b33d5ce541b980/src/SparkFunBME280.cpp#L457) and [pressure](https://github.com/sparkfun/SparkFun_BME280_Arduino_Library/blob/7a7fb985268b7c8ca7678b1658b33d5ce541b980/src/SparkFunBME280.cpp#L391) reference.  It is only initialized in when reading the [temperature](https://github.com/sparkfun/SparkFun_BME280_Arduino_Library/blob/7a7fb985268b7c8ca7678b1658b33d5ce541b980/src/SparkFunBME280.cpp#L490).  This fact should be documented.

I propose adding a single method that reads all three measurements in a single call to the sensor.  In my measurements on a Moteino M0, the time required to read temperature, humidity and pressure can be reduced by about 35% (1139 ns vs 1729 ns).  Reading all values in a single call ensures the sensor values are all from the same measurement.

My first prototype looks like this. However, I am thinking on changing this a bit to align it better with the style of BME280 class with various methods (ie temperature in C / F, altitude in meters / feet).  Thoughts?

```c
struct BME280_SensorMeasurements
{
  public:
	float temperature;
	float pressure;
	float humidity;
};

class BME280
{
  public:
    void read(BME280_SensorMeasurements *measurements);
}
```

santaimpersonator, 2019-12-27:
Thanks for the assist on the other issue. I think your suggestion is valid and convenient if users could specify the measurements they wanted (though, I'm not sure if that the best approach). 

If you submit a pull request, I'd be happy to verify it on a few boards.

cdonate, 2020-12-26:
I've created a PR for this feature. I'm not a C++ guy, so please, check and make any changes you think are necessary. All measurements are read, I don't know if this would be a problem.
#50

nseidle, 2020-12-31:
Thanks @cdonate. Will be in next release.

closed 2020-12-31 by nseidle